hybridgroup/artoo #147: Cannot connect to Arduino UNO through firmata

eskalation, 2015-01-02:
New user here, i am most certainly dealing with an issue on my end, but i have no clue as to what i am doing wrong, so i hope you can help.

I have installed gort, i have uploaded the formata using "gort arduino upload firmata /dev/tty.usbmodem1451".

But whenever i try to connect to my arduino using ruby "ruby blink.rb", it just stops at "Connecting to 'firmata' on port '/dev/tty.usbmodem1451'..."

Please let me know if i am missing something? Thanks!

avermeulen, 2015-02-01:
This is exactly the same issue I am getting - there is now error it just sit there.

I am experienced using firmata with Javascript and Johnny-five - so I am quite sure what I have done so far is correct. Are there any way of getting a more verbose message of why it is just sitting there not doing anything ?

I added some information of what I am trying here:

https://gist.github.com/avermeulen/dd71099312277eb40614

avermeulen, 2015-02-02:
Ok I can report back that the above program, same Arduino board, cable, breadboard setup with LED  is working on Ubuntu 14.04.

But not on OSX 10.8.5
I will play around more and report back.

mrcc87, 2015-08-30:
I Had the same issue and i chaged the serial port. Try using the serial with the cu. prefix instead if the one with tty. I found the hint here: https://github.com/hybridgroup/artoo-arduino

deadprogram, 2015-08-30:
We should probably update the docs to make it easier for OSX users. Many of the examples are currently written assuming Linux port mappings. I think most Linux users can figure out the correct mapping for their configuration vs. many OSX users who might be accessing their ports via code for the first time.

Any thoughts on this?

hunterboerner, 2015-08-30:
> I think most Linux users can figure out the correct mapping for their configuration vs. many OSX users who might be accessing their ports via code for the first time.

Ouch.

But yes, better documentation would be good.

> You might notice that each serial device shows up twice in /dev, once as a tty.* and once as a cu.*. So, what's the difference? Well, TTY devices are for calling into UNIX systems, whereas CU (Call-Up) devices are for calling out from them (eg, modems). We want to call-out from our Mac, so /dev/cu.* is the correct device to use.

> The technical difference is that /dev/tty.* devices will wait (or listen) for DCD (data-carrier-detect), eg, someone calling in, before responding. /dev/cu.* devices do not assert DCD, so they will always connect (respond or succeed) immediately.
>
>   _- http://pbxbook.com/other/mac-tty.html_

watsy0007, 2016-02-14:
thanks @mrcc87 :smile: