hybridgroup/artoo #159: Help me get started.

Sailormanenzo, 2015-08-06:
I need help installing a ruby interpreter and everything else on the Artoo Getting Started Page.

hunterboerner, 2015-08-06:
What operating system are you on?

Sailormanenzo, 2015-08-07:
Mac

hunterboerner, 2015-08-07:
Ok, I use rvm to manage my Ruby installation. You can get it at https://rvm.io/. The basic commands are:

```
gpg --keyserver hkp://keys.gnupg.net --recv-keys 409B6B1796C275462A1703113804BB82D39DC0E3
\curl -sSL https://get.rvm.io | bash -s stable --ruby
```

You may have to install gpg through homebrew (http://brew.sh/).

You can also look at using Cylon.js instead of Artoo if you're more familiar with JavaScript than Ruby. https://github.com/hybridgroup/cylon

Sailormanenzo, 2015-08-07:
Is rvm the interpreter?

hunterboerner, 2015-08-07:
@Sailormanenzo rvm installs and manages different versions of Ruby. If you don't want to deal with that, you could install the latest version of Ruby through Homebrew:

```
brew install ruby
```

Sailormanenzo, 2015-08-08:
Okay thx then what. ( please understand that I am fairly new to this)

hunterboerner, 2015-08-08:
Now you'll need to install artoo. This can be done through rubygems:

    gem install artoo

and then you need to install the specific gem for the platform you want to use. You can see a list of those here: https://github.com/hybridgroup/artoo#hardware-support

Sailormanenzo, 2015-08-08:
So do I type this in a terminal?

hunterboerner, 2015-08-08:
Yes. You also need to install that gem for whichever platform you want to use (e.g. Arduino).

Sailormanenzo, 2015-08-08:
So what's the gem for sphere

Sailormanenzo, 2015-08-08:
Sphero

hunterboerner, 2015-08-08:
"artoo-sphero". You can install it the same way you installed Artoo: `gem install artoo-sphero`.

Here's the project page for examples on how to use Sphero with Artoo: https://github.com/hybridgroup/artoo-sphero

Sailormanenzo, 2015-08-09:
Thx I will let you know when I need more help

hunterboerner, 2015-08-09:
No problem. I'm going to mark this as closed for now. Let me know when you need more help.

closed 2015-08-09 by hunterboerner