hybridgroup/artoo #164: Question: Raspberry Pi and Sensors

epipheus, 2016-06-25:
So the GPIO documentation for the Raspberry Pi does not include sensors? Why not? Does the sensor interface not work for RPi?

deadprogram, 2016-10-04:
Hi @epipheus the Raspberry Pi does not support analog I/O (AIO), due to not having the required hardware interface.

Hope that helps!

epipheus, 2016-10-04:
Wait I don't get why the dsl you've built can't include sensors through the GPIO+ADC

deadprogram, 2016-10-05:
Sorry, perhaps I was unclear. The Raspi has digital I/O but not analog I/O, due to not having an onboard ADC.

epipheus, 2016-10-05:
So does the sensor dsl work with RPi if you add an ADC?

epipheus, 2016-10-06:
Dude, I just want to know if artoo is useful for the case where one wants to use RPi and analog sensors through a 3rd party ADC -- will your below DSL work in this case:

```ruby
  device :sensor, driver: :analog_sensor, pin: 0, interval: 0.25, upper: 900, lower: 200
```

epipheus, 2016-10-06:
I appreciate that you are responding.  But I first asked about this in Jun.

deadprogram, 2016-10-06:
You would need to create a driver for that ADC that consumes the i2c interface for the Raspi, and to provide the AIO interface functions for other devices. Did you have a specific ADC in mind?

epipheus, 2016-10-06:
Anything easily purchased :)

epipheus, 2016-10-06:
Also and again, I'm trying to evaluate using artoo so I don't know. Does
the sensor Dsl work at all for digital sensors or does it not because each
digital sensor simply has its own ADC on board which would need a driver as
you described?

deadprogram, 2016-10-06:
Not sure if you've seen these docs yet: http://artoo.io/documentation/guides/what_is_artoo/ and http://artoo.io/documentation/guides/concepts/ should be helpful.

Basically, in Artoo terms, the raspi is an adaptor that provides the interface to the raspi's GPIO & I2C, but not AIO due to not having any built-in ADC. Artoo's Drivers consume the various interfaces provided by Adaptors, such as GPIO, AIO, or I2C.

What I am working on in Gobot actually, is the idea of a Driver that consumes the I2C interface of the Adaptor for the system it is running on like the Raspi, and then itself provides the AIO interface to other Drivers, so you can easily use an ADC daughterboard such as the RaspIO AnalogZero http://rasp.io/analogzero/

Once I have it working in Gobot, I intend to cross-port it to Artoo.

Anyhow, hopefully this explanation was useful. Thanks for asking good questions.